Aceinna/gnss-ins-sim #2: Error encountered demo_ins_loose

ravindraji4, 2019-04-05:
Dear all, 

Thank you very much for sharing & contributing the open source implementation. 

in function ins_algo_manager()

def run_algo
at line 95 where in code:
results[self.output_alloc[i][j]][this_algo_name+'_'+str(key)] = this_results[j]

this_results is list with value [NoneType]

IndexError: list index out of range as j goes from 0 to 4

how can this error could be corrected??
Thanks for your help in advance.

dxg-aceinna, 2019-04-08:
Thank you for your feedback.

This algorithm is not completed yet. I only set up the framework of the algorithm. Sorry for that.

If you want to try a loosely-coupled GNSS/INS algorithm, please run demo_aceinna-ins.py instead. This demos how to run .dll algorithms on Windows platforms, while demo_aceinna_vg.py (a dynamic tilt sensor algorithm) demos how to run .so algorithms on linux-like platforms.  To make maintainance easier, the .C source code of the mentioned algorithms are not included in this repo. You can get the source code from our OpenIMU platform:
https://www.aceinna.com/inertial-systems/OpenIMU300ZA
Its user manuall is available at:
https://openimu.readthedocs.io/en/latest/

If you want to know how to automatically compile .C files into a .so lib, please refer to the magnetometer calibration  algorithm demo: demo_mag_cal.py. This demos how to manage a .C project using CMake. There is no such demo on Windows platforms. You need manually generate the .dll files.

ravindraji4, 2019-04-08:
Thank you very much for your explanation and hints. 

I would like also to bring to your attention. in demo_free_integration_openimu
the file "ini.txt" is provided from the log_dir, it is missing in the implementation. 

Could it is possible to upload the example files for nxp and bosch??

Thank you again.
Best regards.

dxg-aceinna, 2019-04-09:
demo_free_integration_openimu was uploaded by mistake. This file is used for my own purpose. But since it is already in the repo, I will provide the data file.
Thank you for your feedback.

Best regards.

closed 2019-04-10 by dxg-aceinna