Aceinna/gnss-ins-sim #3: UnboundLocalError: local variable 'mobility' referenced before assignment

Lishen1, 2019-04-16:
def __parse_mode(self, mode):
        '''
        Parse mode. Not completely implemented yet.
        Args:
            mode: simualtion mode
        '''
        if mode is not None:
            if isinstance(mode, str):               # choose built-in mode
                mode = mode.lower()
                if 'flight' in mode:
                    mobility = high_mobility
                elif 'land' in mode:
                    mobility = high_mobility
                elif 'ship' in mode:
                    mobility = high_mobility
                else:
                    mobility = high_mobility
            elif isinstance(mode, np.ndarray):      # customize the sim mode
                if mode.shape == (3,):
                    mobility[0] = mode[0]
                    mobility[1] = mode[1] * attitude.D2R
                    mobility[2] = mode[2] * attitude.D2R
                else:
                    raise TypeError('mode should be of size (3,)')
            else:
                raise TypeError('mode should be a string or a numpy array of size (3,)')
        else:
            mobility = high_mobility
        return mobility

varible mobility not defined

dxg-aceinna, 2019-04-17:
could you please provide more details on how to repeat this error?
1. os
2. version of python
3. python script

Thanks.

Lishen1, 2019-04-17:
I think this is very imortant feature and shoul be integrated as soon as possible becouse without this feature you can not really debug fusion algoritms. If implemented functionality works (i already use this feature and looks that it's works), it will be enough in most cases. BTW, i use this sim antively now, so maby a will have suggestions.  Would be good if velocity data will be simulated.

dxg-aceinna, 2019-04-17:
Indeed there are two kinds of velocity data. One is GPS velocity, of course, and the other one is odometer measurement (velocity expressed in the body frame, for most vehicles, there is only along-track velocity). You can find the details of generating velocities in pathgen.py.
But I did not include odo measurement in ins_sim because:
1. you can indeed recover this infomation via true velocity and true attitude. 
2. I am not familiar with odo and currently working on some other projects. But if you are familiar with odo, please do a pull request to upload your odo model and help us to improve the simulator. 

Thank you again for you feedback. If you have any issue or advice in future, please feel free to let me know.

closed 2019-04-17 by dxg-aceinna