Aceinna/gnss-ins-sim #5: Constants in geoparams

theophilec, 2019-09-05:
Hi!

In the context of IMU simulation in a static setting, we've run into issues using the constants in geoparams because of the "hard-coded" values `FLATTENING` and `E_SQR`.

I'd change to `FLATTENING = 1/298.257223563` and `E_SQR = ECCENTRICITY**2`? 
The first is the WGS-84 definition of flattening. The second removes a 1e-12 error that integrates to small drifts.

Thanks for open-sourcing this simulator: it's great!

dxg-aceinna, 2019-09-06:
Hi theophilec,

Thank you for your feedback.
I chose "hard-coded" values instead of your way to save one multiplication.  In embeded software dev, this is popular.

Could you tell me why such a small "error" matters in your application? Usually this error is considered (ignorable) numerical error and is overshadowed by other error from IMU and GNSS. Indeed, different numerical integration method can cause error much larger.

Dong

theophilec, 2019-09-30:
Hi Dong, 

Sorry for taking so much time to get back to you!
It's just that if we consider the WGS-84 standard in our geo model in our filter, it generates a model mismatch that integrates into something meaningful I found (we are using high-grade IMUs).

I'll try to revert our fork and show you a plot in the coming days!

dxg-aceinna, 2019-10-08:
Hi theophilec,

I replaced  the hard-coded values with WGS-84 definitions. And I will close this issue.
Thank you.

Dong

closed 2019-10-08 by dxg-aceinna