Aceinna/gnss-ins-sim #8: GPS error model dictionary modified in gps_gen

theophilec, 2019-11-19:
I believe there is an issue in GPS noise generation when multiple runs are used, because of the way the `gps_error` dictionary is used.

## Bug reproduction
Using the `demo_no_algo.py`:
1. comment out the plotting line
2. change `sim.run(1)` to `sim.run(10)` (for example)
3. change `ref_frame` from `1` to `0`

Open `gps-0.csv`, `gps-4.csv` and `ref_pos.csv`. As you can see, the `gps-4.csv` data is identical to the `ref_pos.csv` data in the first two columns (lattitude and longitude). However, in the `gps-0.csv` file, this is not the case, and the noise level is consistent with the GPS error model dictionary (here, the built-in one).

## Bug description
In the `gps_gen` function below, the `gps_err` dictionary is a class attribute, passed as reference, that is modified at each simulation loop. And so, dividing by the Earth's radii, becomes exponentially small as runs are simulated.
```
def gps_gen(ref_gps, gps_err, gps_type=0):
    '''
    Add error to true GPS data according to GPS receiver error parameters
    Args:
        ref_gps: If gps_type is 0, [Lat, Lon, Alt, vx, vy, vz], [rad, rad, m].
                 If gps_type is 1, [x, y, z, vx, vy, vz], [m, m, m].
                 ref_gps data are expressed in the navigation frame.
        gps_err: GPS reeceiver parameters.
            'stdp': RMS position error, [m, m, m].
            'stdv': RMS velocity error, [m/s, m/s, m/s].
        gps_type: GPS data type.
            0: default, position is in the form of [Lat, Lon, Alt], rad, m
            1: position is in the form of [x, y, z], m
    Returns:
        gps_mea: ref_gps with error.
    '''
    # total data count
    n = ref_gps.shape[0]
    # If position is in the form of LLA, convert gps_err['stdp'] to LLA error
    if gps_type == 0:   # GPS is in the form of LLA, stdp meter to rad
        earth_param = geoparams.geo_param(ref_gps[0, 1:4])
        gps_err['stdp'][0] = gps_err['stdp'][0] / earth_param[0]
        gps_err['stdp'][1] = gps_err['stdp'][1] / earth_param[1] / earth_param[4]
    ## simulate GPS error
    pos_noise = gps_err['stdp'] * np.random.randn(n, 3)
    vel_noise = gps_err['stdv'] * np.random.randn(n, 3)
    gps_mea = np.hstack([ref_gps[:, 0:3] + pos_noise,
                         ref_gps[:, 3:6] + vel_noise])
    return gps_mea
```

## Fixing the bug
I would like to propose to submit a PR calculating the latitude and longitude standard deviations in local variables. Would you be open to that?

## Environment
Cloned `master` branch in new `Python` `venv` with only `numpy` and `matplotlib` added.

dxg-aceinna, 2019-11-20:
Thank you for your feedback. I fixed this bug. If you have a more elegant solution, please feel free to submit a PR. We always appreciate that.

theophilec, 2019-11-20:
Your fix looks great. Thanks for fixing.

theophilec, 2019-11-20:
Actually, I just noticed that you left a `print` statement in the fix, I imagine it was unintentional.

dxg-aceinna, 2019-11-21:
Yes, it was unintentional, and I just removed it.

closed 2019-11-21 by theophilec