Aceinna/gnss-ins-sim #10: Acceleration error

charlie34, 2020-02-07:
When I get the trayectory and I obtain the accelerometer data and reference quaternion,  I've checked the acceleration measurements from the transformation the acceleration measurement. When the quaternion is [1,0,0,0], in this case, the acceleration measurements is [7.328724153415264002e-03,-7.097695153575984614e-04,-9.801918481487374990e+00].
After the transformation, with the quaternion [0.793 +0.609i +0.000j +0.000k] , the acceleration should be [ 7.32872415e-03  9.46751078e+00 -2.53847319e+00], but the acceleration obtained from simulation is [-0.004411866173716657, -9.473102208586964, -2.5396297352216073]. 
Thanks a lot!
Carlos

dxg-aceinna, 2020-02-07:
1. you need to notice that the reference frame for attitude in the simulator is NED (north-east-down).
2. The quaternion in the simulation represents the rotation from the reference frame to the body frame.
So, for a quaternion [0.793 +0.609i +0.000j +0.000k], you can get the coordinate transformation matrix c_n_to_b = 
[1.0000         0         0
  0    0.2580    0.9661
  0   -0.9661    0.2580]
The accel in the body frame = c_n_to_b * f_NED = c_n_to_b * [0;0;-1]g = [0; -0.9661; -0.2580]g.
You can get the same results using quat_conj x [0;f_NED] x quat.

Thanks for your feedback.

charlie34, 2020-02-07:
The coordinate transformation matrix c_n_to_b is depend on my latitud, long
and alt or is it always the same matrix?
How to calculate the coordinate transformation matrix or how do I obtain it?
thanks a lot!!!

dxg-aceinna, 2020-02-07:
c_n_to_b does not depent on latitude or longitude. I think you mean c_n_to_e. "e" represents ECEF.
c_n_to_b is not a constant matrix. It defines the rotation between the NED frame and the IMU body frame which is usually fixed to some kind of vehicle.
Coordinate transformation matrix can be calculated from quaternion or Euler angles. You can find this in any books on inertial navigation. Actually, you can also find the implementation of such calculation in gnss-ins-sim. In attitude.py, please refer to quat2dcm and euler2dcm.

Dong

charlie34, 2020-02-07:
A last question, why can not you get the acceleration with quaternion
transformation? You should get the same results with DCM and quaternion,
because if you transform the quaternion in DCM, doesn't you do the same?
Thanks for all!! It is a great help!
Carlos!

dxg-aceinna, 2020-02-07:
Please take a look at my first reply. There is the equation of how to do coordinate transformation using quaternion.

charlie34, 2020-02-07:
Now it's good!, but still i have a question. The acceleration measurement
is [-0.007615941414335841, -9.476121803031624, -2.532879477909879] , and
what i've obtained it's
[0.0007232745713307572, -9.457954854321455, -2.533622316334459] , the
x-axis difference is due to noisy, isn,t it?
Thanks a lot.

dxg-aceinna, 2020-02-09:
Sorry for the late reply. It should be dut to noise. The generatation of IMU data has been well tested by different methods.

closed 2020-02-10 by dxg-aceinna