Aceinna/gnss-ins-sim #11: Mahony Algorithm

bababash, 2020-02-11:
Hi @dxg-aceinna! Firstly,this library is super useful!! Thank you very much for putting something so comprehensive together!

I have been looking into your Mahony algorithm and was wondering if there was a source you based it on as it is different from the standard implementations in literature. It almost seems like you merged in elements of the Valenti complementary filter. I would be very interested if you had some relevant articles I could read.

 Thanks!

dxg-aceinna, 2020-02-12:
Hi bababash,
Thanks for your feedback.

The implementation of this complementary filter seems easy. However, its theory is quite complicated. You can refer to "Non-linear complementary filters on the special orthogonal group".

Implementations in other programming languages can be foud here: https://x-io.co.uk/open-source-imu-and-ahrs-algorithms/

My implementation is from a dynamic tilt sensor project I worked on several years ago. I also referred to the above article and source code.

Dong

bababash, 2020-02-12:
Cheers Dong,

Do you have any advice for setting the thresholds for Ki and Kp to switch from low to high? At the moment it's set to 0.2 for both the accelerometer and the gyro. Is there a method you used to determine this number?

Also can you tell me about the `innovationLimit` variable? I have never seen that before in any of the other complementary filters and don't know how to tune it.

Thanks again

dxg-aceinna, 2020-02-13:
Hi bababash,

First, this is only a demo. So, only some simple strategies are implemented. Actually, this threshold is quite difficult to set if you just use the difference between accel measurement and Earth gravity. Let's assume the unint is level, its accel output should close to [0, 0, -9.8]m/s/s. If there is a linear accel along the z axis, the norm of the accel output is norm([0, 0, -10.0]) = 10.0m/s. If there is a linear accel along the x axis, the norm of the accel output is norm([0.2, 0, -9.8]) = 9.802m/s. You can not switch to low gain if you only use the norm of the accel measurement. Further, in this case, a linear accel along the z axis does not affect pitch and roll, while a linear accel along x/y does. You may want to set this threshold to a small value, which may keep the gain always low if there is noise and vibration.

Things are similar to innovationLimit. This is used to limit the effects of high linear accel (vibration maybe). Other implementation of the complementary filter may be just an implementation of the theory. If you want make a product, that is not enough.

Dong

closed 2020-02-14 by dxg-aceinna