Aceinna/gnss-ins-sim #13: Strange value in "lat/long to meters"

junwangcas, 2020-02-26:
Hi,

Function： In the function geoparams.py "lla2ecef(lla)"
Data: motion_def-3d.csv, I modify the init lat/long to 1° and 1°.
Function return value: (x,y) = (6.38E+06 , 1.11E+05).  Actually you can check in the result txt file: ref_gps.csv.
Thanks very much.

Best Regards
Jun Wang

junwangcas, 2020-02-26:
By the way, I run the code: demo_no_algo.py

dxg-aceinna, 2020-02-27:
Hi,
Thanks for your feedback.
You need to first read either readme or the doc of ins.Sim to see the definition of the reference frame. In this demo, the reference frame ref_frame is 1. If you want position in [latitude longitude  altitude] format, you need to set ref_frame to be 0.

junwangcas, 2020-02-29:
Hi Dong,

Thanks very much. I think I can close this issue now.

Jun

closed 2020-02-29 by junwangcas