Aceinna/gnss-ins-sim #19: bias_drift uses wrong std value?

ace-e4s, 2020-09-24:
First, thanks for the amazing tool!

But, I suspect there is a bug here: https://github.com/Aceinna/gnss-ins-sim/blob/2d2ba0fac193afd10b4d7758cc85d14457f17001/gnss_ins_sim/pathgen/pathgen.py#L552

Should it be

```
b = 1/sqrt(fs) * drift[i]
```

or am I missing something?

dxg-aceinna, 2020-09-25:
Hi,

Here I only use a first-order discretization of the first-order Gauss-Markov model: x_dot = (-1/corr_time)*x + n, where n is the driving noise. And you are right. Different sampling frequency will result in different output noise. 

This model is to approxiamte the long-term drift of the gyro bias. Back to the time I built this simulation, I couldn't find a good model of bias drift because drift is really complicated. So I chose two models (first-order Gauss-Markov and first-order random walk) that are commonly used in an INS solution based on Kalman filters.

Actually, this model cannot even precisely model in-run bias-instability. You can check this with demo_allan.py. So, in practice, you can tune the bias_stability config in the simulation to make the long-term drift of bias to be on the same order of that of the sensor spec.

BTW, could you tell me how you get your equation?

ace-e4s, 2020-09-29:
Ohh, I see. Thanks for the answer.

I have no issues with the 1st order Markov-Gauss model. It is used everywhere in the literature and appears to be sufficient for most engineering cases. Although, I think the discretization is "wrong". Because, you are actually discretizing x_dot = (-1/corr_time)*x + dW/dt, where dW/dt is "derivative" of the Brownian motion/Wiener process. (In fact, it is kinda the "anti-integral, since the Brownian motion is nowhere-differentiable.)

Furthermore, dW -> W_(t) - W_(t-1) -> N(0, sqrt(delta_t)).

Anyway, I will give you more detailed description later this week maybe. And I can make a pull-request when I get the time, and you can have a look.

ace-e4s, 2020-09-29:
> BTW, could you tell me how you get your equation?

Its the continuous-time solution to the long-term variance. The discrete version look a bit different (which is the correct one to use in this case), but I'll revert to that as well.

ace-e4s, 2020-10-01:
Let me try again:

We want to discretize the following 1st order Gauss-Markov model:

 `dx/dt = (-1/corr_time)*x + sigma * dW/dt`

where dW/dt is "derivative" of the Brownian motion/Wiener process. Accordingly,

`x_k = (1 - 1/corr_time)*x_(k-1) + sigma * nu_k` 

where `nu_k ~ N(0, dt)`. More conveniently,

`x_k = (1 - 1/corr_time)*x_(k-1) + drift_mark * e_k`

where `e_k ~ N(0, 1)` and `drift_mark = sigma * sqrt(dt)`. Now, `drift_mark` is the driving force. However, this value has limited value in practice since (i) it depends on the 1st order Gauss-Markov assumption, and (ii) it is not the value we derive from Allan Variance analysis.

What we can estimate from Allan variance analysis is the correlation time and the "overall" noise process variance, denoted `drift`. Therefore, we need to relate `drift_mark` and `drift`. The relationship can be derived by studying AR(1) process:

`drift = drift_mark / sqrt(1 - exp(-2/(corr_time * fs)))`

Accordingly, the final discretized model becomes:

`x_k = (1 - 1/corr_time)*x_(k-1) + drift * sqrt(1 - exp(-2/(corr_time * fs))) * e_k`

So, what have we achieved?

1. We can use the values we obtain from Allen variance analysis directly
2. The bias drift process can be assigned an "overall" process variance, i.e. long-term variance.
3. The bias drift process variance becomes independent of the sampling rate, i.e. we don't need to update `drift` parameter when we change the sampling rate.

If you don't mind, I would like to create a pull request on this.

I have created some examples and summary in a notebook you can find here: https://github.com/ace-e4s/noise_error_models
Furthermore, see here for error models: https://www.ncbi.nlm.nih.gov/pmc/articles/PMC3812568/

Wikipedia reading list:

- https://en.wikipedia.org/wiki/Euler%E2%80%93Maruyama_method
- https://en.wikipedia.org/wiki/Autoregressive_model
- https://en.wikipedia.org/wiki/Wiener_process
- https://en.wikipedia.org/wiki/Brownian_motion
- https://en.wikipedia.org/wiki/Ornstein%E2%80%93Uhlenbeck_process 

Disclaimer: This is not my field of expertise.

dxg-aceinna, 2020-10-13:
Hi, thank you for the detailed explanation. I carefully read the materials you provided. But I still cannot get a clear picture of the theory.

First, there are some minor mistakes.

1.  >x_k = (1 - 1/corr_time)*x_(k-1) + sigma * nu_k
    
      should be

     >x_k = (1 - dt/corr_time)*x_(k-1) + sigma * nu_k

2. >nu_k ~ N(0, dt)

    should be

    >nu_k ~ N(0, dt^2)

3. >drift_mark = sigma * sqrt(dt)

    should be

    >drift_mark = sigma * dt

4.  >drift = drift_mark / sqrt(1 - exp(-2/(corr_time * fs)))

    To understand this one, I read your notebook.

$\sigma_{x}^2 = \frac{\sigma^2 \Delta t}{1 - e^{-2\beta\Delta t}} = \frac{\sigma_{rw}^2}{1 - e^{-2\beta\Delta t}}$

ace-e4s, 2020-10-19:
> Second, the key equation is not well explained.
> 
> > drift = drift_mark / sqrt(1 - exp(-2/(corr_time * fs)))
> 
> To understand this one, I read your notebook. But I cannot get a clear answer. Why can drift and drift_mark be related this way? A detailed derivation is appreciated. Thank you.

No problem! I'll get back to you as soon as possible...

ace-e4s, 2020-10-21:
I'll try to answer your comments using the same numbering:

1. You are right, this is a typo from my part.
2. No, the variance is `dt`. See https://en.wikipedia.org/wiki/Euler%E2%80%93Maruyama_method (last sentence in the introduction).
3. Follows 2.

ace-e4s, 2020-10-21:
> Second, the key equation is not well explained.
> 
> > drift = drift_mark / sqrt(1 - exp(-2/(corr_time * fs)))
> 
> To understand this one, I read your notebook. But I cannot get a clear answer. Why can drift and drift_mark be related this way?

I found this relationship in https://www.ncbi.nlm.nih.gov/pmc/articles/PMC3812568/ (Eq. 3).

However, I think this is a numerically stable (very good) approximation. The actual relationship should look something like this:

First, realize that the discretized first order Gauss-Markov model is indeed a AR(1) model. (Autoregressive model):

`x_(k) = (1 - dt/corr_time)*x_(k-1) + drift_mark * e_k`

Now, `drift = VAR[x_(k)]`. And for an AR(1) model the the variance is given as:

`drift = drift_mark / (1 - (1 - dt/corr_time)**2)` (See https://en.wikipedia.org/wiki/Autoregressive_model Section "Example: An AR(1) process").
 
Note that `(1 - dt/corr_time)**2` is the Taylor expansion of `1 - exp(-2 * dt/corr_time)` upto the second order. Also, note that dt/corr_time must be less than or equal to unity to make physical sense. (corr_time > dt or the whole discretization kinda falls a part.)

I think they use `1 - exp(-2 * dt/corr_time)` because `exp(x) - 1` can be calculated with good accuracy for small x, e.g.  https://docs.scipy.org/doc/scipy/reference/generated/scipy.special.expm1.html#scipy.special.expm1

I hope this was helpful.

dxg-aceinna, 2020-10-22:
> 
> 
> I'll try to answer your comments using the same numbering:
> 
>     1. You are right, this is a typo from my part.
> 
>     2. No, the variance is `dt`. See https://en.wikipedia.org/wiki/Euler%E2%80%93Maruyama_method (last sentence in the introduction).
> 
>     3. Follows 2.

Thank you very much. I think I understand this one now.

dxg-aceinna, 2020-10-22:
> Note that `(1 - dt/corr_time)**2` is the Taylor expansion of `1 - exp(-2 * dt/corr_time)` upto the second order. Also, note that dt/corr_time must be less than or equal to unity to make physical sense. (corr_time > dt or the whole discretization kinda falls a part.)

Thank you for the explanation. It is clear to me now.
A pull request will be much appreciated.

PR #26 (2021-02-20): Fix #19
fix #19
commit 84c67c30 (2021-02-20): Fix #19 (#26)
closed 2021-02-20 by dxg-aceinna