Aceinna/gnss-ins-sim #20: self.amgr.run_algo error

ravindraji4, 2020-09-29:
Hi, 
Thanks for the open source algorithm, almost all functioned for me. 
I would like bring to your kind attention the following error in **demo_ins_loose.py**

`Traceback (most recent call last):
  File "C:/gnss_ins_sim/demo_ins_loose.py", line 63, in <module>
    test_ins_loose()
  File "C:/gnss_ins_sim/demo_ins_loose.py", line 55, in test_ins_loose
    sim.run(100)
  File "C:\gnss_ins_sim\gnss_ins_sim\sim\ins_sim.py", line 177, in run
    algo_output = self.amgr.run_algo(algo_input, range(self.sim_count))
  File "C:\gnss_ins_sim\gnss_ins_sim\sim\ins_algo_manager.py", line 101, in run_algo
    results[self.output_alloc[i][j]][this_algo_name+'_'+str(key)] = this_results[j]
IndexError: list index out of range`

Thank you very much for your help in advance!!

dxg-aceinna, 2020-09-30:
Hi,

I have never been able to find enough time for a python-version INS algorithm. You may use an older version of gnss-ins-sim now. For the latest version, this demo will suggest you try demo_aceinna_ins.py. However, demo_aceinna_ins.py use a dll version of the algorithm will providing the C source code. To get the C source code, you can refer to the source code of demo_aceinna_ins.py for details.

Thank you for your feedback.

ravindraji4, 2020-09-30:
Thank you for your reply. 
So that means that this algorithm is embedded inside the dll version of  **demo_aceinna_ins.py**?

dxg-aceinna, 2020-10-09:
You need to first install VS code. In VS code, you install Aceinna extension, and you can get access to the source code via Aceinna extension.

For more details, please refer to:
https://openimu.readthedocs.io/en/latest/install.html
You don't need the ST Link driver.

closed 2020-10-13 by dxg-aceinna