Aceinna/gnss-ins-sim #22: Generated KML files does not output altitudes <0

elcf, 2021-01-27:
I'm not sure if this was done on purpose, but if an altitude is <0 (either reference or algo), the KML altitude is set at 0. The output pos CSV files however state the negative altitudes.

elcf, 2021-01-27:
For the issue that the KML has positive altitude data, but in Google Earth Pro it shows as 0, the reason is that the default "altitudeMode" is set to "clampToGround" Source: https://developers.google.com/kml/documentation/kmlreference#gx:altitudemode
So it seems it has to be made explicit that the "altitudeMode" is "absolute".

elcf, 2021-01-27:
Saw that in line 122-123 of kml_gen.py, `if lla[i][2] < 0: lla[i][2] = 0`. Is there a reason this was done? I will change it on my code since I'm using it for underwater sim.

dxg-aceinna, 2021-01-28:
Hi elcf,

Thank you for your feedback. This was done on purpose. When there is negative altitude and the altitude mode is "absolute", this point is not showed on Google Earth because it might be underground. So I forced the altitude to be non-negative.
[image]
[image]

elcf, 2021-01-30:
Hi dxg, understood, thank you.

closed 2021-01-30 by elcf