Aceinna/gnss-ins-sim #24: Running demo_no_algo.py out of the box returns NotImplementedError

Chris2L, 2021-02-02:
Downloaded repo
Opened powershell
ran `py -m venv env`
ran `& env/Scripts/Activate.ps1`
ran `pip.exe install numpy`
ran `pip.exe install matplotlib`
ran `python.exe .\demo_no_algo.py`

Got the following error
NotImplementedError: Axes3D currently only supports the aspect argument 'auto'. You passed in 'equal'.

Quick look in code did not show where we set it as equal or where to set it as auto

Chris2L, 2021-02-02:
line 201 in sim_data_plot.py had to change to `axis = fig.add_subplot(111, projection='3d', aspect='auto')`

dxg-aceinna, 2021-02-03:
Hi Chris2L,

Thank you for your feedback.
The reason why I put "equal" there is that I want the XYZ ratio to be 1:1:1 to better represent the trajectory in 3D plot. Back to the time I wrote the code, "equal" is not implemented for "3D" but matplotlib does not rais an error, either. I was hoping this function could be implemented in some version. I seems that matplotlib chose to raise an error instead of implementing this function.
You can check this issue here:
https://github.com/matplotlib/matplotlib/issues/1077/

I will see if there is a workaround.

dxg-aceinna, 2021-02-03:
Hi Chris2L,

I got a workaround using set_box_aspect. Please refer to: https://github.com/matplotlib/matplotlib/issues/17172.
However, set_box_aspect is not available in older versions of matplotlib. So, I won't fix this now. If you are interested, you can use the following code:
```
    axis = fig.gca(projection='3d')
    axis.set_box_aspect((1, 1, 1))
    # axis = fig.add_subplot(111, projection='3d', aspect='equal')
```

Chris2L, 2021-02-03:
That works great.  Thanks

closed 2021-02-03 by Chris2L