Aceinna/gnss-ins-sim #29: how to simulate a magnetometer

gaopenggang, 2021-04-19:
I am trying to simulate an imu augmented magnetometer based on a motion trajectory. if I can know the magnetic strength and robot pose, I can calculate the mag_x, mag_y,mag_z. But how can I know the magnetic strength? I can do it using GPS, but, GPS is not available in our situation.
Or could you recommend some papers or tutorials on how to simulate a magnetometer?

dxg-aceinna, 2021-04-19:
I am not quite clear about your question. However, I will try to answer that. If you find my answer is not for your question, please add more details.

1. You want to simulate a magnetometer-realted algorithm. In this algorithm, you will also use IMU. The algorithm is something like an AHRS algorithm.
2. To simulate the algorithm, you need to generate the magnetometer measurement.
3.  To generate the magnetometer measurement, you need to know the geomagnetic field. Besides strength, you also need to know declination and inclination. Please refer to http://autoquad.org/wiki/wiki/configuring-autoquad-flightcontroller/autoquad-calibrations/calibration-faq/magnetic-declination-and-inclination/
4. To know strength, declination and inclination, you need a model to calculate these three parameters. The model I use in gnss-ins-sim is from https://github.com/cmweiss/geomag/tree/master/geomag
5. Actually, I don't think the model really matters. The main problem is how you deal with different kinds of magnetic interferences in your algorithm.

gaopenggang, 2021-04-19:
Thanks a lot.  How to know the To know strength, declination, and inclination?  If I have a GPS, I can get it by `https://github.com/cmweiss/geomag/tree/master/geomag`.
If I do not have a GPS, I can know the geomagnetic field through some websites after input its location.  For example, Edinburgh, UK. has a field with a magnitude of 50 microTesla, a 70 degree downwards inclination, and 0 degrees declination.
Right?

dxg-aceinna, 2021-04-20:
For simulation the magetometer measurement, you can assume the locaiton is known.
However, in your algorihtm, you might have to assume that you can never know the strength, declination or inclination because there are so many anomalies in the world where the local geomagnetic field does not follow the math model.

closed 2021-04-26 by dxg-aceinna