Aceinna/gnss-ins-sim #32: GPS results always zero whatever the motion profile and algorithms

ovasseur, 2022-01-13:
I am running simulations and I cannot make GPS results to be other than 0, except for the altitude which varies if I make the vz velocity to vary.

I have tried a lot of different motion profiles:
- Various initial states (zero and non-zero velocities; zero and non-zero Euler angles);
- Various motion commands (zero and non-zero changes in Euler angles or velocities; different command types);
- I made sure I have `gps=True` in the IMU model and that GPS visibility is 1;
- I tried with no algorithms and with different built-in algorithms;
- I tried with the two reference frames.

In all my experiments, GPS latitude and longitude (`gps_lat`, `gps_lon`) stayed at there initial value. I am refering to both the 'ref_gps' results and the 'gps-0' results (with or without GPS uncertainty).
I was expecting the latitude and longitude to vary when for example vx or vy vary, which is not the case.

In addition, in the results with uncertainties, only the altitude has noise, while latitude and longitude are always flat curves.

To reproduce this issue, I am attaching the results I obtain with the motion file `motion_def-simple.csv` (also attached) and with the main code `demo_no_algo.py`.

[accel-0.csv](https://github.com/Aceinna/gnss-ins-sim/files/7864892/accel-0.csv)
[att_euler.csv](https://github.com/Aceinna/gnss-ins-sim/files/7864893/att_euler.csv)
[image]
[att_quat.csv](https://github.com/Aceinna/gnss-ins-sim/files/7864894/att_quat.csv)
[gps-0.csv](https://github.com/Aceinna/gnss-ins-sim/files/7864895/gps-0.csv)
[image]
[gps_time.csv](https://github.com/Aceinna/gnss-ins-sim/files/7864896/gps_time.csv)
[gps_visibility.csv](https://github.com/Aceinna/gnss-ins-sim/files/7864897/gps_visibility.csv)
[image]
[gyro-0.csv](https://github.com/Aceinna/gnss-ins-sim/files/7864898/gyro-0.csv)
[ref_accel.csv](https://github.com/Aceinna/gnss-ins-sim/files/7864899/ref_accel.csv)
[ref_att_euler.csv](https://github.com/Aceinna/gnss-ins-sim/files/7864900/ref_att_euler.csv)
[image]
[ref_att_quat.csv](https://github.com/Aceinna/gnss-ins-sim/files/7864901/ref_att_quat.csv)
[ref_gps.csv](https://github.com/Aceinna/gnss-ins-sim/files/7864902/ref_gps.csv)
[image]
[ref_gyro.csv](https://github.com/Aceinna/gnss-ins-sim/files/7864903/ref_gyro.csv)
[ref_pos.csv](https://github.com/Aceinna/gnss-ins-sim/files/7864904/ref_pos.csv)
[image]
[ref_vel.csv](https://github.com/Aceinna/gnss-ins-sim/files/7864905/ref_vel.csv)
[summary.txt](https://github.com/Aceinna/gnss-ins-sim/files/7864906/summary.txt)
[time.csv](https://github.com/Aceinna/gnss-ins-sim/files/7864907/time.csv)
[motion_def-simple.csv](https://github.com/Aceinna/gnss-ins-sim/files/7864922/motion_def-simple.csv)

dxg-aceinna, 2022-01-17:
Hi ovasseur,
Thank you for the feedback.

I checked the ref_gps.csv. Both longitude and latitude vary although the changes have very small values, which is reasonable.
You can use the following simplified formula to roughly convert the latitude changes in deg to distance in meters.
distance_in_meter = lat_change_in_deg / 180 * pi * ( earth_radius+altitude )
Earth radius can be 6378137m.

To get a better accuracy, you need to use a real Earth model (WGS84, for example).

ovasseur, 2022-01-17:
> Hi ovasseur, Thank you for the feedback.
>
> I checked the ref_gps.csv. Both longitude and latitude vary although the changes have very small values, which is reasonable. You can use the following [...]

Hi dxg,

Thank you very much for your answer!

closed 2022-01-17 by ovasseur