Aceinna/gnss-ins-sim #36: Why does ‘g’ need to be multiplied by -1 in the NED coordinate system?

WongRZ, 2022-09-27:
Dear author,
In #10 Issue, you mentioned that " The accel in the body frame = c_n_to_b * f_NED = c_n_to_b * [0;0;-1]g ". 
My question is, why does g need to be multiplied by -1 in the NED coordinate system? Isn't the direction of gravity consistent with that of D?
Thanks a lot.

dxg-aceinna, 2022-10-11:
Hi WongRZ,

Sorry for the late reply because I haven't been working on this project for a while.
I hope you have found the answer. But if not, please let me explain.

First, the accelerometer measure NON-GRAVITATIONAL acceleration:
f = a - g
f is the accelerometer measurement vector, a is acceleration we can observe, and g is gravity vector.

When the unit is not moving, a = [0; 0; 0].
On Earth, in the NED frame, g = [0; 0; 1]g.

So, f = -g = [0; 0; -1]g.

WongRZ, 2022-10-13:
Thanks a lot for your reply.
I found the answer, which is consistent with your reply. And the simulation data works in my code now.
This is a very convenient simulation project. Thank you for your efforts.

dxg-aceinna, 2022-10-14:
Thank you for your feedback.

closed 2022-10-14 by dxg-aceinna