Aceinna/gnss-ins-sim #38: A question about GPS measurement

lixiangpengcs, 2022-12-01:
During the simulation, I found a problem. After stdp and stdv are specified, the simulated GPS height may tend to a certain direction of the true value. For example, the average GPS position height of the track is about -1.5m according to the statistics of the simulated GPS data. Even though I didn't do any settings on the height. Is this normal?

dxg-aceinna, 2022-12-02:
I tested this with the following script. From the logged file gps_0.csv, the height has an average of about 0m.
Could you give more details to reproduce the issue you have?

`   
    ####
    #### choose a built-in IMU model, typical for IMU381
    imu_err = 'mid-accuracy'
    gps_err = {'stdp': np.array([2.5, 2.5, 2.5]),
               'stdv': np.array([1.0, 1.0, 1.0])}
    # generate GPS and magnetometer data
    imu = imu_model.IMU(accuracy=imu_err, axis=9, gps=True, gps_opt=gps_err)
    
    #### start simulation
    sim = ins_sim.Sim([fs, fs_gps, fs_mag],
                      motion_def_path+"//motion_def-static.csv",
                      ref_frame=0,
                      imu=imu,
                      mode=None,
                      env=None
                      algorithm=None)
    sim.run(1)
`

lixiangpengcs, 2022-12-02:
Sorry. It is a mistake by my program. There is no problem with this simulation.

closed 2022-12-02 by lixiangpengcs