Aceinna/gnss-ins-sim #40: How to create high freq trajectory from low freq trajectory.

fghoussen, 2023-04-07:
Looking for an example to create high freq trajectory from low freq trajectory.

The most similar example to start from seemed to me to be demo_free_integration.

Running the demo_free_integration.py example produces empty plots.
I modified  demo_free_integration.py this way:
```
>> git diff demo_free_integration.py
diff --git a/demo_free_integration.py b/demo_free_integration.py
index fc2e804..b7274eb 100644
--- a/demo_free_integration.py
+++ b/demo_free_integration.py
@@ -58,8 +58,7 @@ def test_free_integration():
     ini_pos_vel_att[3:6] += ini_vel_err
     ini_pos_vel_att[6:9] += ini_att_err * D2R
     # create the algorith object
-    algo1 = free_integration_odo.FreeIntegration(ini_pos_vel_att)
-    algo2 = free_integration.FreeIntegration(ini_pos_vel_att)
+    algo = free_integration.FreeIntegration(ini_pos_vel_att)
 
     #### start simulation
     sim = ins_sim.Sim([fs, 0.0, 0.0],
@@ -68,14 +67,14 @@ def test_free_integration():
                       imu=imu,
                       mode=None,
                       env=None,
-                      algorithm=[algo1, algo2])
+                      algorithm=[algo])
     # run the simulation for 1000 times
-    sim.run(10)
+    sim.run(1)
     # generate simulation results, summary
     # do not save data since the simulation runs for 1000 times and generates too many results
     sim.results(err_stats_start=-1, gen_kml=True)
     # plot postion error
-    # sim.plot(['pos'], opt={'pos':'error'})
+    sim.plot(['ref_pos', 'pos'], opt={'ref_pos': '3d', 'pos':'3d'})
 
 if __name__ == '__main__':
     test_free_integration()

```

This gives empty plots:

[image]

Reading the doc, I tried to simplify the case:

```
>> git diff demo_motion_def_files/motion_def-90deg_turn.csv
diff --git a/demo_motion_def_files/motion_def-90deg_turn.csv b/demo_motion_def_files/motion_def-90deg_turn.csv
index 87caf81..12a3b3a 100644
--- a/demo_motion_def_files/motion_def-90deg_turn.csv
+++ b/demo_motion_def_files/motion_def-90deg_turn.csv
@@ -1,6 +1,4 @@
 ini lat (deg),ini lon (deg),ini alt (m),ini vx_body (m/s),ini vy_body (m/s),ini vz_body (m/s),ini yaw (deg),ini pitch (deg),ini roll (deg)
 31.9965,120.004,0,10,0,0,315,0,0
 command type,yaw (deg),pitch (deg),roll (deg),vx_body (m/s),vy_body (m/s),vz_body (m/s),command duration (s),GPS visibility
-1,0,0,0,0,0,0,1,0
-1,15,0,0,0,0,0,6,0
-1,0,0,0,0,0,0,3,0
+1,0,0,0,0,0,0,10,1

``` 

And still get empty plots.

What did I miss?

My initial need is to create high freq trajectory from low freq trajectory: what is the most similar example? Any advice on how to achieve this would be helpful.

Note: my low freq trajectory is a collection of n-uplets {timestamp, lat, lon, alt, roll, pitch, yaw, ax, ay, az, wx, wy, wz}

dxg-aceinna, 2023-04-11:
command type of -1 is not supported.

To create high freq trajectory from low freq trajectory, I think you can choose command type of 1. You need to calculate the body frame velocity from two adjacent positions, and the Euler angel change rate from two adjacent attitudes, and then generate the motion profile file. However, the generated accel and angular rate cannot perfectly match those in the low frequency trajectory, I think.

fghoussen, 2023-04-12:
> command type of -1 is not supported.

The `-1` is part of `git-diff` output: lines begin with `1` and `-` / `+` means the lline has been suppressed / added.
The file is:

```
>> cat demo_motion_def_files/motion_def-90deg_turn.csv
ini lat (deg),ini lon (deg),ini alt (m),ini vx_body (m/s),ini vy_body (m/s),ini vz_body (m/s),ini yaw (deg),ini pitch (deg),ini roll (deg)
31.9965,120.004,0,10,0,0,315,0,0
command type,yaw (deg),pitch (deg),roll (deg),vx_body (m/s),vy_body (m/s),vz_body (m/s),command duration (s),GPS visibility
1,0,0,0,0,0,0,10,1

```

My understanding is that with these files, I should start from the position (31.9965,120.004,0) with velocity (10,0,0). Then mode 1 should make the trajectory evolves as this "Euler angles change rate and body frame velocity change rate". So I should get a trajectory in the matplotlib viewer AFAIU.

What did I miss ?

dxg-aceinna, 2023-04-13:
I tried your code, and got results as expected.

Maybe you can provide the output of the script.

------------------------------------------------------------
Sample frequency of IMU: [fs] = 100.0 Hz
Reference frame: 1
Simulation time duration: 10.0 s
Simulation runs: 1

dxg-aceinna, 2023-04-19:
It seems the position, velocity and attitude are generated. Could you also run the built-in demo demo_no_algo.py to see if the 3D plot is empty or not?

fghoussen, 2023-04-19:
Here is what I get:

[image]

fghoussen, 2023-04-20:
I have a matplotlib install which seems to work

[image]

dxg-aceinna, 2023-04-21:
I think you can check if data files are generated and saved in your local disks. If they are, I think the problem is with matplotlib. I did have a problem with different versions of matplotlib. If 3D plots matters, you can modify the code in sim_data_plot.py.

dxg-aceinna, 2023-04-21:
You can also disable 3D plot by removing `opt={'ref_pos': '3d'}` to see if 2D plot is available.

fghoussen, 2023-04-22:
>  if data files are generated and saved in your local disks. 

Where are data saved? What are the names of the data files?

dxg-aceinna, 2023-04-23:
In `ins_sim.py`, read the doc for
`def results(self, data_dir=None, err_stats_start=0, gen_kml=False, extra_opt='')`

closed 2023-04-27 by dxg-aceinna