Aceinna/gnss-ins-sim #41: Pathgen.py calculation of the gyro output

Chris2L, 2023-04-25:
I've been stuck on this problem for a few weeks now.  I'm developing an error state Kalman filter and using the output of this sim as test data.  
It never estimates the errors 100% correctly, so I started running my prediction code only, figuring that, if using the ref data, I should get the same answer.  I've reduced all variables to only now update my attitude using the gyro reading.
I set ref_frame = 1, so not taking into account earth's rotation etc (which I do calculate exactly the same as the sim).  In calc_true_sensor_output I add the following code to test the output of the gyro.
`    q1 = attitude.euler2quat(att)
    print(q1)
    q2 = attitude.quat_update(q1, gyro, 0.5)
    print(q2)
    print("_______________________")
    global count
    count += 1
    if count == 3:
        quit()`
with a count in the global frame.  Basically what I'm trying to do is get the current quaternion, update it with the gyro rate, and print the new quaternion. (Yes I did hardcode dt at 0.5s, and this is also the dt used as input, the error is bigger at larger dt's).  My motion file is basically, start at 0, 0, 0 with vel 0,0,0 and attitude 0,0,0.  Have the following command

> 1,0,0.3,0.3,0,0,0,1200,1
so flipping 360 deg in pitch and roll at the same time.

I get the following printout:

> [1. 0. 0. 0.]
> [9.99999983e-01 1.30899693e-04 1.30899693e-04 0.00000000e+00]
> _______________________
> [ 9.99999983e-01  1.30899692e-04  1.30899692e-04 -1.71347298e-08]
> [ 9.99999856e-01  3.79609089e-04  3.79609089e-04 -8.22467001e-08]
> _______________________
> [ 9.99999856e-01  3.79609076e-04  3.79609076e-04 -1.44103071e-07]
> [ 9.99999461e-01  7.34347081e-04  7.34347081e-04 -4.13426700e-07]
> _______________________

So the first is the quat where we start and the second is the quat where we end after applying the gyro rate.  
The third is the quat calculated from the att, and that should in theory be the same as the second correct?  But it is not, even by a few fractions.  In my ESKF I get the same value as the second, so my calculations are done the same as the built-in attitude.quat_update.

I'm I doing something wrong?  Is my assumption about how to apply the rate command wrong?  Is this just a limitation of double precision?

dxg-aceinna, 2023-04-26:
Let me first double confirm what problem you have.
1. The problem is "The third is the quat calculated from the att, and that should in theory be the same as the second correct? But it is not, even by a few fractions."
2. In your problem, att is Euler angles converted from the second quat, and att is then converted back to get the third quat. In theory, the second quat and the third quat represent the same attitude, and should be the same.

If I understand your problem correctly,
1. this problem has nothing to do with quat udpate, right?
2. this should be due to numerical accuracy.

Chris2L, 2023-04-26:
Thank you for the quick response.
I've expanded my test to more clearly articulate my problem.  I've added the following code:

> att2 = attitude.euler_update_zyx(att, gyro, 0.5)
> print(f"att({count})  = {att[0]:.12f}\t{att[1]:.12f}\t{att[2]:.12f}\t")
> print(f"att'({count+1}) = {att2[0]:.12f}\t{att2[1]:.12f}\t{att2[2]:.12f}\t")

I then get this output
> att(0)  = 0.000000000000        0.000000000000        0.000000000000
> att'(1) = 0.000000000000        0.000261799388        0.000261799388
> 
> quat(0)  = 1.000000000000       0.000000000000        0.000000000000        0.000000000000
> quat'(1) = 0.999999982865       0.000130899693        0.000130899693        0.000000000000
> _______________________
> att(1)  = 0.000000000000        0.000261799388        0.000261799388
> att'(2) = 0.000000000000        0.000759218225        0.000759218225
> 
> quat(1)  = 0.999999982865       0.000130899692        0.000130899692        -0.000000017135
> quat'(2) = 0.999999855897       0.000379609089        0.000379609089        -0.000000082247
> _______________________

So att(t) is the attitude fed into the function and is the reference.  att'(t+1) is then what the att should be in the future given gyro reading.  You will see that att'(1) and att(1) match up 100%.  This is what I would expect.
quat(0) is att(0) converted to quat, so should be the reference.  I then rotate quat(0) given gyro to get quat'(1), but this does not match quat(1).  
From your comment, you believe this is due to numerical inaccuracy correct?  I had a feeling it might be, but hoped that because quaternions are the best thing since sliced bread, it should not be so susceptible.  I've also tried a Python package called mpmath to try and increase the floating point accuracy but did not get close to accurate enough.
Is it worthwhile to change my code to use Euler updates in most cases and then switch over to quaternion updates when you get close to gimbal lock?

dxg-aceinna, 2023-04-27:
I think I understand your problem. That is, attitude propagation results from Euler update and quaternion update are different.
That is as expected.

Attitude propagation has different forms which are described in many articles, papers or textbooks, Strapdown analytics (Savage) for example. Using gyro to generate a rotation quaternion and then applying this rotation quaternion to current quaternion to get future quaternion assume constant angular rate between current and future. That is a simplification. Euler angle update in the simulation tool is an even simplified method. It is a first order simplification of numerical integration.

I would suggest using quaternion update, and considering sculling depending on your application.

The reason why I use Euler update is, in the simulation, gyro rates are generated from Euler angles, and in the free integration demo, using Euler angle update (in a reverse way of generating gyro rates from Euler angles) can produce results better matching the reference values. It's for demo only. In practice, I see few people using Euler update.

Chris2L, 2023-05-02:
Thank you for the feedback!  It does help a lot!  
You would suggest something like this:
[image]
for [this](https://www.vectornav.com/resources/inertial-navigation-primer/math-fundamentals/math-coning) site correct?

Using the simulator output I got closer results using the following code:
```
def quat_update(q, w, dt):
    # rot_quat = rotation_quat(w, dt)
    delta_theta = w*dt

    rot_quat = euler2quat(delta_theta[::-1])
    q = quat_multiply(q, rot_quat)
    q = quat_normalize(q)
    return q
```
This is what you are referring to in your comment :
> Using gyro to generate a rotation quaternion and then applying this rotation quaternion to current quaternion to get future quaternion assumes constant angular rate between current and future.

To confirm, you would propose a final system to rather use the first mentioned update cycle even though it does not follow the euler update from the simulator, it would still give a better result using actual data?
Thank you for the feedback in understanding the problem!

dxg-aceinna, 2023-05-06:
Yes for the following question. This is true especially when the angular rate is high and there is rotation for all three axes. And in practice, people may choose to use a higher-order solution depending on applications. For INS on automotive, I see people using coning  compensation.

> To confirm, you would propose a final system to rather use the first mentioned update cycle even though it does not follow the euler update from the simulator, it would still give a better result using actual data?

closed 2023-05-08 by Chris2L