Aceinna/gnss-ins-sim #50: Simulate attitude change of IMU sensor

oniht, 2024-06-05:
Hello, is it possible to simulate the attitude change of the IMU sensor without changing the attitude and velocity of the vehicle? 

For example, I want to simulate that the IMU data would come from a smartphone inside the vehicle and that the attitude of the smartphone could change during the ride.

Thanks!
Oskari

dxg-aceinna, 2024-06-06:
The attitude defined in the simulation is between the IMU and a local reference frame (NED). It cannot simulate the attitude between the IMU and the vehicle.
In your use case, if the attitude change between the IMU and the NED cannot be easily modeled/defined, it will be difficult to simulate the motion of the IMU.

oniht, 2024-06-06:
Thanks for the quick reply.

If I want to simulate an acceleration in the reference frame y-direction (east) such that the IMU is inclined by certain roll (e.g. 20deg) and pitch (20deg), can I do it simply by rotating the acceleration (or velocity) vector to the inclined IMU frame. 

For example if I want to achieve vy_ned = 8 m/s in 10 sec. Final velocity vector in NED frame would be v = [0, 8, 0] and in rotated IMU frame v = [0, 7.52, -2.74]. 

So would it work by giving this motion definition?

| Ini lat (deg) | ini lon (deg) | ini alt (m) | ini vx_body (m/s) | ini vy_body (m/s) | ini vz_body (m/s) | ini yaw (deg) | ini pitch (deg) | ini roll (deg) |
|---|---|---|---|---|---|---|---|---|
| 32 | 120 | 0 | 0 | 0 | 0 | 0 | 20 | 20 |
| command type | yaw (deg) | pitch (deg) | roll (deg) | vx_body (m/s) | vy_body (m/s) | vz_body (m/s) | command duration (s) |	GPS visibility |
| 1 | 0 | 0 | 0 | 0 | 0 | 0 | 200 | 1 |
| 5| 0 | 0 | 0 | 0 | 7.52 | -2.74 | 10 | 1 |

dxg-aceinna, 2024-06-06:
Yes, I think it will work.

closed 2024-06-17 by dxg-aceinna