Aceinna/gnss-ins-sim #51: Is there a plan for demo_ins_loose to be completed?

layssi, 2024-06-20:
It would make the simulation much more realistic.

dxg-aceinna, 2024-06-24:
Hi layssi, thanks for asking. However, I don't have the bandwidth to write a python-based INS algorithm now.

As an alternative, you can refer to demo_aceinna-ins.py for an INS algorithm compiled into a dynamic library. A few years ago, aceinna open sourced one of its INS algorithms. As to how to access the source code of this algorithm, you need to install vscode, to install the aceinna extension inside vscode, and then to get a INS project inside the aceinna extension. In the project, you can find the INS algorithm.  But you still need to write some wrapper files to get the algorithm to fit into the simulation.

Now, the INS algorithm in aceinna is no longer open-source. So, I can share the source code now.

layssi, 2024-07-08:
OK. Thanks

layssi, 2024-07-08:
OK. Thank you

closed 2024-07-08 by layssi