Aceinna/gnss-ins-sim #53: How to simulate Human Handwriting Dynamics

vkosuri, 2024-12-20:
I am currently working on a project that involves modelling and recognizing human handwriting dynamics using IMU sensor data. 

The sensors in my setup include accelerometers, gyroscopes, magnetometers, and force sensors embedded in a digital pen. 

My objective is to generate realistic handwriting trajectories and simulate noisy IMU data for training and validating handwriting recognition algorithms.

While reviewing the `imu_err` configuration, I found the following example:  

```python
imu_err = {
    'gyro_b': np.array([0.0, 0.0, 0.0]),  # Gyro bias, deg/hr
    'gyro_arw': np.array([0.25, 0.25, 0.25]),  # Gyro angle random walk, deg/rt-hr
    'gyro_b_stability': np.array([3.5, 3.5, 3.5]),  # Gyro bias instability, deg/hr
    'gyro_b_corr': np.array([100.0, 100.0, 100.0]),  # Gyro bias instability correlation, sec.
    'accel_b': np.array([0.0e-3, 0.0e-3, 0.0e-3]),  # Accelerometer bias, m/s^2
    'accel_vrw': np.array([0.03119, 0.03009, 0.04779]),  # Accel velocity random walk, m/s/rt-hr
    'accel_b_stability': np.array([4.29e-5, 5.72e-5, 8.02e-5]),  # Accel bias instability, m/s^2
    'accel_b_corr': np.array([200.0, 200.0, 200.0]),  # Accel bias instability correlation, sec.
    'mag_std': np.array([0.2, 0.2, 0.2])  # Magnetometer noise std, uT
}
```

My questions are:  
1. Can this noise model be directly adapted for simulating fine motor movements like handwriting dynamics?  
2. Are there specific adjustments or parameter ranges you would recommend to better match the characteristics of human handwriting (e.g., lower velocities, high-frequency micro-movements)?  
3. How can I customize this `imu_err` model to incorporate tilt and force dynamics alongside trajectory generation?  
4. Is there an example where GNSS-INS-SIM has been used for non-GNSS-specific trajectories, such as object motion or handwriting-like patterns?  

Thank you for providing such a versatile tool, and I look forward to any suggestions you can share to help refine this approach.

dxg-aceinna, 2024-12-20:
To my understanding, to add noise to an nominal IMU signal (signal without error) is pretty simple because we mathematically model IMU noise by differnt types. The difficulty for your application is to model the physical motion (nomial IMU) of hand writing. Using the motion profile, we can define the physical motion. However, it requires a lot of lines to define the motion of hand writing. You can see from the demo motion profiles that, even for cars, we can only roughly model their physical motion.

There is another problem you need to consider. In this simulation, the motion is defined at the location of the IMU. For your application, you want to define the motion at the pen point and the IMU is very unlikely to be located at the pen point. The lever arm between the pen point and the IMU can be easily modelled, though.

vkosuri, 2024-12-21:
Thank you for the detailed explanation and for highlighting the key challenges. I appreciate your insights, and I’d like to address the points raised while seeking further guidance:

> To my understanding, adding noise to a nominal IMU signal (signal without error) is pretty simple because we mathematically model IMU noise by different types. The difficulty for your application is to model the physical motion (nominal IMU) of handwriting. We can define the physical motion using the motion profile. However, defining handwriting motion requires many lines. You can see from the demo motion profiles that, even for cars, we can only roughly model their physical motion.

I understand that defining precise handwriting motion profiles is inherently more complex than modeling vehicle dynamics due to fine motor control and variability in handwriting strokes. To address this challenge:

- **Leveraging Handwriting Datasets**:  
   We have access to the [OnHW dataset](https://www.iis.fraunhofer.de/de/ff/lv/dataanalytics/anwproj/schreibtrainer/onhw-dataset.html), which contains IMU motion data for handwriting dynamics. This dataset includes:
   - **Accelerometer Data**: Acc1 X, Acc1 Y, Acc1 Z (front accelerometer) and Acc2 X, Acc2 Y, Acc2 Z (rear accelerometer) values in three dimensions.
   - **Gyroscope Data**: Gyro X, Gyro Y, Gyro Z values in three dimensions.
   - **Magnetometer Data**: Mag X, Mag Y, Mag Z values in three dimensions.
   - **Force Data**: The force with which the pen tip touches the surface.
   
   By extracting motion trajectories for individual characters from this data, we can create realistic motion profiles for simulation.

   - Would it be feasible to integrate real trajectories from datasets like this into GNSS-INS-SIM to simplify defining the nominal IMU motion?

- **Automating Motion Profiles**:  
   For a single character (e.g., "A"), we could use parametric models (e.g., Bézier curves or splines) to approximate strokes. This reduces the need to manually define every detail and allows interpolation for smoother paths. Here's an example:
   ```python
   # Example for "A"
   trajectory_points = [
       (0, 0), (0, 2),  # Upward stroke
       (0, 2), (1, 0),  # Downward diagonal
       (0.3, 1), (0.7, 1)  # Horizontal middle stroke
   ]
   ```
   Could such parametric approximations work as a starting point in GNSS-INS-SIM?

> There is another problem you need to consider. In this simulation, the motion is defined at the location of the IMU. For your application, you want to define the motion at the pen point and the IMU is very unlikely to be located at the pen point. The lever arm between the pen point and the IMU can be easily modeled, though.

You’ve rightly pointed out the challenge of simulating motion at the pen tip due to its offset from the IMU. To address this:

- **Lever Arm Correction**:  
   The offset can be modeled using rigid-body dynamics. The motion at the pen tip can be calculated as:

   $$\text{Pen Tip Motion} = \text{IMU Motion} + R(\theta) \cdot \text{Lever Arm}$$

   where \( R(\theta) \) is the rotation matrix derived from IMU orientation.  
   Implementing this correction would ensure that pen-tip motion reflects the writing trajectory accurately.  
   Would you be able to recommend any specific approach or resources within GNSS-INS-SIM to facilitate this?

- **Combining Dataset Trajectories and Lever Arm Correction**:  
   By applying lever-arm transformations to the real dataset's IMU data, we could simulate pen-tip trajectories more accurately, integrating them back into the GNSS-INS-SIM framework.

dxg-aceinna, 2024-12-23:
>       * Would it be feasible to integrate real trajectories from datasets like this into GNSS-INS-SIM to simplify defining the nominal IMU motion?
The problem I had when I started this simulation tool is should I start with a position trajectory and then get velocity and acceleration by calculating derivatives. In this way, a smooth trajectory is required. Otherwise, there will be large variation in its derivatives especially when delta time is very small (very high sampling rate): velocity = delta_position / delta_time. So I ended up with the current motion definition. It is not very convenient but can avoid very large linear or rotational accelerations. So, to your question, I think it is possible in theory. However you need to verify it.

>       Could such parametric approximations work as a starting point in GNSS-INS-SIM?
Same as above. In this case, the trajectory shall be smooth.

>       Implementing this correction would ensure that pen-tip motion reflects the writing trajectory accurately.
>       Would you be able to recommend any specific approach or resources within GNSS-INS-SIM to facilitate this?
Please make sure that the rotation matrix and its derivatives are known.
GNSS-INS-SIM provides some basic funtions for coordiante transformation. However, without detailed derivaiton of equations in your application, I don't know if such functions will help much.

dxg-aceinna, 2024-12-23:
You can also takle a look at http://www.instk.org/

I referred to this tool when I started GNSS-INS-SIM.

mgeorge-is, 2025-04-21:
Hello, I built www.inertialsim.com to support both synthetic (simulated) and log data inputs.  Some of the functionality is similar to what you were looking for here. Any combination of position, orientation, velocity, and/or acceleration and angular rate can be the input.  Simulated IMU signals are the output.  All numerical derivatives and gravity/earth rate effects will be taken care of internally.  Take a look if it might work for your application.